hoglet67/RGBtoHDMI #416: Atari VBXE RGBs advice

brad-colbert, 2025-02-02:
Hello,

I'm looking for settings advice to properly display the output from a VBXE video enhancement board for the Atari XL/XE.

The output is RGBs @15kHz.  The output is 320x240@256 colors

The only profile that captures it is CGA.   I've played with some of the settings to create a custom profile but I'm not able to get better than this color resolution.

IanSB, 2025-02-03:
@brad-colbert 

A brief search indicates that the VBXE is an FPGA based video enhancement board outputting component video. Is that correct?

What type of RGBtoHDMI are you trying to use with this? (analog / digital etc)

The analog version will not work as it can only capture 2 bits each of RGB and it looks like the VBXE outputs 7 bits each of RGB. (The analog version doesn't use a normal A to D but does thresholding instead which results in noise free video but will only work with a few levels of video)

You might have better luck connecting the digital version of RGBtoHDMI to the digital inputs of the DAC on the VBXE but even then that would only be 4 bits each of RGB.

Currently the best option for the Atari is the c0pperdragon lumacode board which encodes the video in such a way that it can be output on the RF connector as a baseband mono signal which can be decoded into a bit perfect HDMI signal by RGBtoHDMI.

Maybe the VBXE developer could add a similar lumacode style option?
https://github.com/c0pperdragon/LumaCode/wiki/GTIAdigitizer-(for-Atari-8bit)

brad-colbert, 2025-03-02:
My apologies for not replying.  Thank you so much for looking into this.  I fully understand now that it's not a proper use of the RGBtoHDMI and thank you for a suggested solution.

Cheers!