hoglet67/RGBtoHDMI #417: Excessive flicker and ghosting when using Lumacode 2600 and 5200

nick3092, 2025-02-02:
I recently installed C0pperdragon's Lumacode boards in a 2600 and 5200.  While everything seems to work well for the most part, some games do exhibit excessive flicker and/or ghosting in certain cases on both systems.  I also noticed that when I open the RGBtoHDMI menu, the objects that flicker will sometimes disappear and reappear as I move through the menu.  Not knowing if the issue was in the lumacode device or RGBtoHDMI, I started out by reaching out to C0pperdragon.  He explained that a programming trick used on older system to get more objects on the screen is to render an object every other frame.  And he explained that while in the menu, RGB to HDMI locks on to every other frame as you navigate the menu, which explains why the objects appear and disappear.  It appears those objects drawn every other frame are the ones that exhibit this behavior.  He suggested I ask about the issue here to see if anything can be done on the RGBtoHDMI end to improve the image.

One of the easiest games to replicate the issue with is Kangaroo for the 2600.  when your character climbs up to the second level on the first screen, the strawberry fruits and and your character begin to flicker, and when moving the kangaroo has ghosting.  The monkeys that come down to this level to throw things at you also exhibit this behavior.  The effect can also be seen in Yars' Revenge on the first level as well.  The shield around the enemy base has a slight flicker to it.  And once you chip away some of the shield, it begins ghosting as it moves up and down.

I took a few pictures of the issue.  The Kangaroo ghosting was a particularly bad one, I think it may have been exaggerated based on when the camera exactly captured the picture.  I also included a picture of the RF output (which is particularly ugly, probably because all the shielding is off currently) where you can see in the RF signal, instead of flickering or ghosting, affected objects has a comb like effect.  And I also played the game in the Stella emulator, and the objects do have some flicker depending on how many are on the screen, but the flicker doesn't seem to be as bad as through RGBtoHDMI, and there is zero ghosting that I can see.

I am using the special Lumacode RGBtoHDMI with integrated analog from C0pperdragon.  I do have a standard RGBtoHDMI with analog board, but I currently don't have a cable made up to connect to Lumacode with it.  I'm using the Beta 65 image for my SD card, and I have the input on my TV set to PC, and it is invoking game mode automatically.  I have a Retrotink 4k as well, the issue presents both through the RT4k (just using the default HDMI input settings) as well as directly to my TV (2019 Samsung QLED).

TIA for any input on the issue.

[image]

[image]

[image]

IanSB, 2025-02-07:
@nick3092 
I think this is probably due to the way that some TVs and monitors process the incoming video.
e.g. many LCDs try to improve their response times by detecting a significant change in video content between frames and then overdriving the LCD to make that change more rapid. However this technique is not really designed for continuous flickering on/off of images on alternate frames which is what is happening with multiplexed sprites so that overdriving can lead to ghosting or even temporary burnin.
The TV's composite input is probably trying to deinterlace the incoming video (hence the combing) which will tend to blend adjacent frames together so it has fewer problems with the overdriving.

nick3092, 2025-02-10:
Thanks for the reply @IanSB Is there anything that can be done within the firmware to better deinterlace the video or something to at least make it more like the RF comb or even the way Stella renders it seemingly more solid but dimmer?  The ghosting and flickering is quite annoying and is really distracting depending on the game.

IanSB, 2025-02-11:
@nick3092 
I might be able to try blending frames or something like that but here is one other possibility to try:

[1920x1080_Interlace.zip](https://github.com/user-attachments/files/18751469/1920x1080_Interlace.zip)

Unzip the above and put the 60Hz file into /resolutions/60Hz on the SD card. Similarly put the 50Hz one into /resolutions/50Hz 

Then reboot the Pi zero and manually select the 1920x1080_Interlaced resolution in the main menu.
This will output a 1080i signal which your TV might try to deinterlace thus forcing it to do some processing on the video.
Note that the aspect ratio will be messed up with the image appearing narrow as the software will need some tweaking to properly support this mode but it should be enough to see if it makes any difference. (If it does make a difference I will make those software changes.)

nick3092, 2025-02-12:
Thanks so much for taking a look at this.  The interlaced resolution produces an image pretty much identical to how RF input looks on my TV.  The objects that are drawn every other frame now have a comb look to them.  It makes the Harmony flash cart menu look really well too, as it too uses the every other frame trick to draw the letters of the menu.  In Kangaroo, its generally OK looking.  But in Yars, where you have large blocks of those alternating pixels constantly moving up and down, the leading edge looks almost solid now instead of ghosting.  In the attached screenshot, the enemy shield was moving down.  So the leading edge of the pixels on the bottom look odd.

I hooked up my Coleco Gemini as the RF is amazingly clear on that compared to an actual 2600, and I can see the same thing happening via RF with the leading edge of the pixels in Yars' if I look closely.  Its just much more apparent when using Lumacode/R2H, as the screen is so much brighter and crisp.

Do you think a frame blending solution would give an even better appearance?  Looking closer at the Stella emulator (which does a a very good job of displaying these types of objects as solid and not ghosting or comb) I think they are doing some kind of blending under the name of a "phosphor" setting.  Looks like it must have some kind of detection or internal database of ROMs, as the phosphor setting is either "by ROM" or you can manually set it via a slider.

[image]

IanSB, 2025-02-12:
@nick3092 
Try this alpha release which should give you the correct aspect ratio:

[Alpha66A.zip](https://github.com/user-attachments/files/18769809/Alpha66A.zip)
Wipe the SD card to install and select the 1920x1080(Interlaced) resolution.

Some form of phosphor emulation / blending might produce better results but that will take some time to look at.

nick3092, 2025-02-12:
Thanks, that update did fix the aspect ratio for the interlaced mode.  And I totally understand that any kind of blending/"phosphor" system would take some time to implement (if it's even possible).  If you get around to something like that, just ping me and I'll gladly test it for you.