hoglet67/RGBtoHDMI #418: Can't get a stable picture with NS405 based video card (MFA Microcomputer)

hanshuebner, 2025-02-07:
Hi,

I'm trying to use a rgb2hdmi with the 6-12 bit analog RGB board to capture and display the output of an NS405 based terminal card for the MFA Microcomputer.  Following the instructions, I have set the pixel clock to the 18.181818 Mhz stated in the NS405 data sheet and tried to adjust the sample and geometry settings, but I am unable to get a completely stable picture.  Here is a little video that shows the display instability: https://youtu.be/xSTXlyKJk7E

Does anyone recognize this type of problem and know how to fix it?

Thanks in advance!
Hans

IanSB, 2025-02-09:
@hanshuebner 

There are several possibilities:
1. You have the wrong pixel clock. Can you confirm there is an 18.1818 Mhz crystal on the board?
2. The video signal is being chopped up by the pixel clock which means that the clock setting and corresponding H pixel dimensions have to be doubled to sample properly.
3. The signal you are picking up isn't at full logic levels and is a lower voltage. The data sheet for that chip shows it has an intensity bit which can be used with external circuitry to reduce the video level. (This can be worked around by picking up the video signal earlier when it is at TTL levels or using an analog RGBtoHDMI which can be set to any arbitrary voltage levels).

Do you have a schematic for the terminal card showing the video path?

hanshuebner, 2025-02-09:
@IanSB Thank you for getting back!  I've had another look at the NS405 data sheet and the schematic, and I found that the pixel clock is equal to the frequency of the crystal that is installed on the board, which is a 12 Mhz part.  I've tried to use exactly that frequency and doubled it, but both attempts resulted in the picture to be multiplied and unreadable.  The best I can seem to get is this:

[image]
[image]

These are also the settings in the video.

The schematics for the video card is this:

[image]

The character cell size on the display output is 7x12 pixels, if I am reading the [datasheet](https://ataribits.weebly.com/uploads/5/8/8/6/58865739/ns405_video_controller.pdf) correctly.  The standard resolution of the output at 80x25 characters is 560x300 pixels, although it is possible to change that to 96x29 (672x348).

I've also connected the video card to a multinorm Sharp LCD TV which is very forgiving and displays most signals that are somewhat TV compatible.  It also displays the NS405's output just fine and appears to be stable, but it is hard to be sure as it has a fairly low resolution.

I'm rather lost, so further instructions would be most appreciated.  Thanks in advance!

hanshuebner, 2025-02-09:
I forgot to include the geometry settings:

[image]

hanshuebner, 2025-02-09:
@IanSB I've wiped the SD Card and installed beta 65.  I've then adjusted the Sync settings to match the profile that I had before so that I could enter the "Create Custom Profile" menu.  There, I set the pixel clock to 12 Mhz and the geometry to 560x300.  With that, the picture is stable and sampled perfectly.  As soon as I leave the setup menu, however, the screen starts flashing between the perfect image and a completely black at about 1 Hz.  Here's my settings so far:

[image]
[image]
[image]

Thank you for helping with this!  It seems I'm getting closer, but your further advice will be much appreciated!

IanSB, 2025-02-09:
@hanshuebner 
Did you save your new custom profile in the custom profile menu and exit / reboot?
Also posting the source summary page from the info menu would be helpful.

hanshuebner, 2025-02-09:
@IanSB I've rebooted a couple of times.  I've tried a couple further adjustments, but the flashing remains.  Here's the source summary page:

[image]

hanshuebner, 2025-02-09:
@IanSB  I've adjusted the line length, but the flashing is still the same:

[image]

IanSB, 2025-02-09:
@hanshuebner 
The height and voffset are too large:
voffset + height = 22 + 300 = 322 but there are only 312 lines in a frame so it is running off the end of the screen

This means that 300 is probably not the right value for the height 
Font size is 7x11 and there are 25 lines = 11x25 = 275 so set Min and Max height to 278 and adjust voffset to centralise that
The total of voffset and height must be less than 312 by at least several lines

hanshuebner, 2025-02-09:
It is coming together!  I've made the adjustments you suggested and also tuned the DAC-A level a little to get rid of some pixels flickering, and now the picture is completely stable.  I will try to display some text with the intensity attribute set to see if that works or if it requires some more tweaking that I'll probably also need some help with.

[image]

hanshuebner, 2025-02-09:
I've just reviewed the documentation of the terminal card firmware and it seems that it does not support intensity.  Thus, I think this can be called done and a complete success.

```
sampling=2,2,2,2,2,2,2,0,7,0,5,0,0,0,0,0,1,8,1,0,57,256,256,256,256,16,256,256
geometry=128,22,560,300,560,300,1,2,0,1,12000000,817,5000,342,4,0,0
palette=RGBrgb_(Amstrad)
scanline_level=0
```

This is the  BFZ/MFA 8.4 video card which is part of the MFA Mikrocomputer system, an educational, modular system from Germany based around the 8085 microprocessor.  The profile should probably be called  "BFZ/MFA 8.4 (80x24 mode)".

IanSB, 2025-02-09:
@hanshuebner 
The profile info will probably have to come from /saved_profiles (assuming you used save configuration after making those changes), not /profiles as the data shows the line length as 817 and max/min as 300 which were the unadjusted values so can you repost that file data from the right folder

It looks like the hardware is configured for intensity support though as that bit is on pin 25 which is connected to the video circuitry

hanshuebner, 2025-02-09:
@IanSB Here's the right file:
```
sampling=0,0,0,0,0,0,0,0,7,0,5,0,0,0,1,0,1,0,1,0,60,256,256,256,256,36,256,256
geometry=164,31,560,278,560,278,1,2,0,1,12000000,770,5000,342,4,0,0
palette=RGBrgb_(Amstrad)
scanline_level=0
```
I've seen that the intensity output is connected, but the terminal firmware does not appear to have support for controlling it.  I'll ask around if someone has another version.  Note sure if I want to venture into patching the 8048 code myself.

Thank you again for your help!  I'm very happy to have this system on my main retro monitor now!

closed 2025-02-09 by hanshuebner

IanSB, 2025-02-09:
@hanshuebner 

Something still not quite right about that as the number of lines per frame in geometry is set to 342 and that should be 312 as shown in the status page. (This was probably set incorrectly during the initial save custom profile because of the error setting the pixel height to 300)
This doesnt actually cause a problem unless there are multiple sub-profiles in a folder to auto select but it should be set correctly and it would have been if the initial pixel height value was correct I think.
I will correct that on the profile I include in the next release.

hanshuebner, 2025-02-09:
@IanSB I've made that adjustment as well and can confirm that the profile still works after I've made that change.

IanSB, 2025-02-09:
@hanshuebner 
If you get intensity support working, you need to make the following changes:

In the Palette menu:
Change the palette to "Mono 3 level bright"
(To get green change Output colour to Green)
In the sampling menu 
Change the Sample mode to "6 bits per pixel"
Change the clock multiplier to x8 (This should be changed anyway as x4 is low for 12Mhz)
Adjust DAC-A G Hi for the bright threshold
Adjust DAC-B G Lo for the dim threshold (will be lower than DAC-A)