hoglet67/RGBtoHDMI #419: Olivetti M19 mono output

goatM19, 2025-02-14:
Hello,

I'm trying to configure the software to catch a greyscale MDA output. I understand some vendors found out how to drive MDA screens via 2 bits to 16 shades, possibly using subpixeling, frame divison, etc. And that this is not documented.

Project currently supports EGA Wonder 800+, which should be using same techinque(s) to drive EGA output on the MDA screen.

The M19 output runs at 640x400@45Hz, with a pixel clock of standard 14.31 MHz.

Using EGA Wonder 800+ second subprofile as template, I've managed to sync up and get 4 out of 16 levels.

[image]

I presume this is just interpreting Intensity and Video as Blue 0 and Green 0 in RrGgBb.

My question would be, does this bundled 800+ profile cover the monochrome target, e.g. point 3 : 

[image]

so a modification of that system could work for M19, or the greyscale MDA thing needs to be implemented first?

Great project btw, happy to be an user
Best regards

goatM19, 2025-02-19:
Hello Ian, thanks for response.

I can provide scope data for M19's output. Current assumption is two bits per clock per line (aligned to both hi and lo of clock).
The mapping of values into 16 shades is also still unknown. 

Is it realistic to ask for this feature? I would also accept pointers about code locations where the additions need to be done, and do it myself.

The basic preset should be EGA since M19's output is best synced to it (400 line 22kHz), but aware of multilevel MDA input.

IanSB, 2025-02-20:
>Current assumption is two bits per clock per line (aligned to both hi and lo of clock).
The mapping of values into 16 shades is also still unknown.

That presumed format is very similar to lumacode which is already supported but uses the G Hi and G Lo bits in analog mode with two or more sub-pixels per original pixel to get digital video out of some retro computers like the 16 colours of the commodore 64 via a single cable carrying an encoded mono signal see here:
https://github.com/c0pperdragon/LumaCode/wiki

You need to create a profile that samples at double rate to see what is going on with the video.
i.e. in the create custom profile menu double the pixel clock, the clock cycles per line and the horizontal pixel resolution and then screencap the video while displaying a test pattern with all 16 colours. Then view that in a paint program to see what is being output.

goatM19, 2025-02-20:
Can I do it via config file, it's faster than pressing buttons...

Current one :

`sampling=6,6,6,6,6,6,6,0,1,0,9,0,0,0,0,0,0,1,0,0,100,256,100,256,100,256,256,256
geometry=12,0,640,400,640,400,1,1,0,1,14318180,768,4500,411,1,0,0
auto_switch=1
palette=RrGgBb_(EGA)
scanline_level=0`

Change geom to 

`geometry=12,0,1280,400,1280,400,1,1,0,1,28636360,1536,4500,411,1,0,0`

IanSB, 2025-02-20:
>Can I do it via config file, it's faster than pressing buttons...

yes, but you will have to double the horizontal offset too (first number 12 -> 24)

goatM19, 2025-02-26:
I am not getting stable sync anymore with doubled values, it's close but not stable.

[image]

https://github.com/user-attachments/assets/709e9578-a33a-491e-8e74-d2a1623da468

The sync is imperfect but a line division appears in all blues. This is a signal with Intensity at 0 and Video "in between" 0 and 1.

Additionally, in EGA Wonder 800+ case there is a software tuning utility for grayscale. Which means the output encoding has variations. The hi TTL output could be delayed for a configurable period. This might be 50/50 relation in some cases. But in some might be 33/66 or vice versa. There is a fair amount of setting levels in that utility - about 8 or more. Each one of those does something different to the pixel signal timing. In M19's case it's a fixed config.

This would make M19 video/intensity signal a PWM which is what oscilloscope suggests too. 

What would be the options in this case?

IanSB, 2025-02-28:
>The sync is imperfect

Overclocking the core by 50 or so may fix that problem

>This would make M19 video/intensity signal a PWM which is what oscilloscope suggests too
What would be the options in this case

If it is actually variable PWM and not simply 50% of a standard pixel width then it is unlikely to work as you would have to sample at some very high multiple of the pixel clock which would not be practical.

However if it is 50% then the diagonal strobing pattern shown above is indictative of mis-sampling caused by incorrect sampling settings so you could try adjusting the line length setting and re-running the auto calibration as described in the wiki

https://github.com/hoglet67/RGBtoHDMI/wiki/Tutorial-on-Adding-a-New-Profile