hoglet67/RGBtoHDMI #422: Wiring directly to the pin header on the board.

Identity-4, 2025-03-12:
It's probably really simple but I cant figure this out:  I have V-Sync, H-Sync, and Video source coming out of the computer (Mac Plus), but where on the pin header on the **12-bit issue 7** board does video source plug into?  There is a pin labeled VSg1, is that V-SYNC, Video Source, or Green, or both?  It would be nice to just have a simple direct wiring diagram.

[image]

IanSB, 2025-03-18:
@Identity-4 
 >There is a pin labeled VSg1, is that V-SYNC, Video Source, or Green, or both?

It's both, normally it's Vsync but it can also double up as g1 if used with the 9bits (V Sync) Sample mode (which requires composite sync on HS). This allows 9bpp capture on the standard 6x2 pin header.

Sorry I haven't put up any Mac wiring info yet but it would be Vsync to VS (Pin 10), Hsync to HS (Pin 8) and video to G3 (Pin 9) with ground on Pin 3 and +5v on Pin 12 of the 12 way IDC connector.

If you want to power it from the Mac do not plug in a USB power supply when doing that as it will back feed +5v into the Mac.

Identity-4, 2025-03-18:
Thanks IanSB, that worked perfectly!  I plan on powering it directly from the power supply, so thanks for the warning on USB power.  I have kind of a strange project going with my Mac Plus, so I am trying to minimize and miniaturize wiring, boards, and peripherals as much as I can on this build (think of a Mac Plus motherboard and OS with fully modern peripherals but running inside a 1U rack - A Rackintosh Plus).

closed 2025-03-22 by Identity-4