hoglet67/RGBtoHDMI #425: Crash when adjusting V-offset

ConventionalMemories, 2025-03-23:
Spent some time creating a profile for my GRiD Compass 1101.
I have everything dialed in pretty well I think. but unfortunatly I can't seem to set the V-offset in in the Geometry menu. As you might be able to see on this picture the top 1 or 2 lines are cut off (border is supposed to be on topp as well)
When trying to adjust the V-offset the pi crashes. (I can move it the other way, but not down)
[image]
Might be something small I've overlooked, but I can't seem to figure it out. 
Running Beta65 on a pi zero. 
GRiD connected to Grn3, HS, VS. 

sampling=1,1,1,1,1,1,1,0,7,0,1,0,0,0,2,0,0,0,0,1,255,100,255,100,255,255,255,256
geometry=72,0,320,240,320,240,1,1,3,1,7500081,464,5000,266,0,0,0
palette=Mono_(2_level)
output_colour=3

ConventionalMemories, 2025-03-23:
I just loaded the latest stable release on a new sd card (20240213), and started from scratch making a new profile.
And unfortunatly I still have the exact same issue, the whole pi crashes when adjusting the V-offset one more line. 
If I change the sync to V+ instead of V- I can see that the top line is beeing captured and that there is some random data infront of the actual pixels. But if I try to center the image with V+ it cuts off the bottom rows instead of crashing. 
(also picture flickers when exiting menu, but that is probably because V sync has to be negative instead of positive)

pictures with sync set to V+ :
[image]

Best I can get with V+, now the bottom rows are cut off, if I change the V-offset it moves the bottom up and does not display more lines below it.
[image]

ConventionalMemories, 2025-03-23:
Some more testing and I think that I require a negative V-offset, which the current build doesn't support. the max value of 256 also isn't enough to fully diplay the next frame.

IanSB, 2025-03-25:
@ConventionalMemories 

As you have discovered, you can't have a negative V offset, when you go below 0 it wraps around to a high value which is probably causing the lockup. When using the opposite polarity and a value of 256 for the offset, you are actually skipping every other frame which is why you get the flashing when exiting the menu.

I suspect that the Vsync pulse you are using is actually a square wave with one edge right at the start of video rather than a proper vsync pulse which normally fits well within the vertical blanking period.

This can occur when picking up video from internal monitor connections which use signals that aren't proper sync signals.
e.g. on the PET 2001 and some other systems the Hsync pulse is actually wider than the blanking period (it is actually a "H Drive" signal for the monitor). The start of the Hdrive signal occurs before the end of active video and the end of it occurs after the start of video so there was no way to use it to trigger video capture. I worked around that by adding an optional delay in the CPLD but there isn't enough space in the CPLD to do something like that for Vsync.

If it is only missing one line at the top of the screen then it's possible that I'm losing that in the transition from vsync to hsync at the start of capture and it might be possible to do something about that in future.

Are there any schematics available?

In the meantime the only workarounds I can suggest are:
1. Look for an alternative vertical signal (e.g. if there is a video out option there might be a proper sync signal available somewhere)
2. Build a circuit to create a proper vsync pulse from the squarewave.

ConventionalMemories, 2025-03-25:
Thanks for taking a look ata this, that sound like it is the problem.
The GRiD Compass only has the interal EL diplay so probably no "normal" sync signal anywhere, also no dedicated CRTC, Scematics are unfortunatly not available. We have been working on reverse engeneering one but afaik nobody has tacheld the video circuitry yet. 

I did however find the datasheet for a similar, if not electrically identical EL display panel. 
Here is a copy of the timing diagram. 

[image]
[image]