hoglet67/RGBtoHDMI #427: miniPET composite out setting

NuClear235, 2025-04-11:
I have miniPET with composite monochrome output. I tried to use RGBtoHDMI mono Lumacode version. Picture is horrible. I tried similar profiles and make my own profile. Picture is too living and strange breathing-characters are partially incomplete. Does not anybody already have working profile for it, please?

IanSB, 2025-04-12:
@NuClear235 
Sorry I don't have an exact profile for the miniPET as I don't have one but you could try this as a starting point:

[Mini_PET_Test.txt](https://github.com/user-attachments/files/19722327/Mini_PET_Test.txt)

Put it in \Profiles\6-12_BIT_RGB_Analog\Commodore on the SD card, reboot and select the profile.

Then post a screenshot of the source summary page in the info menu

NuClear235, 2025-04-13:
I tried Your setting and is EXCELLENT!
Perfect sharp picture. I just set output to green and now is it like original PET.
-I dont know how to insert snap screen here :-(
Thank You

NuClear235, 2025-04-13:
[image]
[image]
[image]
[image]
[image]

IanSB, 2025-04-13:
@NuClear235 
OK it looks like the miniPET video timing is exactly the same as a PET 2001 as that was the timing used in my profile (except using composite sync instead as separate sync)

I'll add that profile to the next release.

NuClear235, 2025-04-14:
OK. My miniPET is only 40 characters! Is called "miniPET".
 
There is next miniPET called "miniPET 40/80D", it has selectable 80 characters per line. This I dont have, so I can not test it.
[https://www.tfw8b.com/product/minipet-40-80-standalone-computer-kit/]

NuClear235, 2025-04-22:
I see strange behaviour-aprox. one of 5 Resets of miniPET shows RGB2HDMI bold fonts.
See pics. One is bold, one is normal.

[image]
[image]

NuClear235, 2025-04-22:
Yes, I tried. Problem is still here.
XTall is this:

[image]

IanSB, 2025-04-27:
@NuClear235 

Did the 16Mhz profile make any difference?

NuClear235, 2025-04-27:
Yes. It helped. Font is never Bold after resets.

But is still a bit wider than before.

[image]

NuClear235, 2025-04-27:
So was before without bold.

[image]

IanSB, 2025-04-27:
@NuClear235 

One other thing to try is adjusting the DAC-A setting in the sampling menu which controls the level slicing that determines the difference between black and white but if that doesn't help then I think this might be a bug in the Mini Pet hardware/firmware which is outputting slightly wider pixels randomly on power up. 

Because RGBtoHDMI samples the middle of a pixel and then paints it over the whole pixel area on the LCD it can amplify the effect so 1.5 pixels would become 2 but I suspect if you feed the miniPET signal to an analog composite monitor you might see a difference in pixel width when you notice the effect on RGBtoHDMI.

NuClear235, 2025-04-27:
I will test it on CRT, If I will see differences after restarts.