hoglet67/RGBtoHDMI #430: problems with vertical synchronization

alex3331, 2025-05-09:
Have two computers - Sharp-MZ-700 and MZ-1500. Until recently had the same Acorn Electron Profile working fine for both. I accidentially changed a setting so that somehow MZ-700 still works OK, but MZ-1500 now seems to have good horizontal sync, but the screen turns blank with ~1 hz frequency.
Both computers connected using RGB+HS cable. No wiring was changed. What can it be? Can provide screenshots.
Regards

IanSB, 2025-05-09:
@alex3331 
>Until recently had the same Acorn Electron Profile working fine for both

Is this a digital or analog RGBtoHDMI?
Can you post a screencap of the source summary page for both systems (in the info menu)

Why are you using the Electron profile? It has the wrong settings for the Sharp MZ-700

There is an actual Sharp MZ-700 profile.

alex3331, 2025-05-09:
Hello, and thanks for answering. This is digital RGBtoHDMI;  After I switched to Acorn Atom it resumed working normally. I did not know that MZ-700  profile now exists; I did not update software for a few years. Let me try to upgrade  it. Hope I will manage it.
Regards

alex3331, 2025-05-09:
Sorry for a dumb question, but how do I add  the MZ-700 profile? Should I just remake SD card with BETA 65?

IanSB, 2025-05-10:
@alex3331 

Yes, wipe the SD card and unzip the latest beta (which is beta66 not beta65)
https://github.com/IanSB/RGBtoHDMI/releases

Note that there is a new way of changing profile. You use the "Select Profile" option at the bottom of the menu, select the computer manufacturer then select the profile.

Note that the MZ-700 profile is setup for PAL timing as that is the system I have. 
An NTSC based machine will need some tweaking to the settings because the pixel clock will probably be different.
Also I wired up the cable for composite sync but the MZ700 also has separate sync out. If you wire up with separate sync you will have to change the Sync setting in geometry (which might be your initial problem)

What is the video pinout for the MZ-1500? Based on info found online, the MZ-800 dropped the composite sync out so has to use separate H/V sync so is that the case with the MZ-1500?

Is there a schematic available?

alex3331, 2025-05-10:
Yes, I have manuals and schematics for MZ-700,  MZ-800  and MZ-1500, as well as all these  computers in working state. I use RGBtoHDMI for MZ-700 and MZ-1500. However, MZ-800 has RGBI output and 16 color palette; I played with a couple homemade  DAC devices with no much luck.
Please let me know what you need and I will provide it to you. 
Alex

IanSB, 2025-05-10:
@alex3331 
>However, MZ-800 has RGBI output and 16 color palette

That will work with RGBtoHDMI with a suitable profile. I can create one for you to test.

First things I need to know
Which of your machines are PAL or NTSC

What are the maximum resolutions and pixel clock frequencies used on those machines. 
If you don't know that then what are the frequencies of the crystals on each of the boards?

e.g. The MZ-700 PAL has a crystal of 17.7344Mhz on the CPU board and the pixel clock is 8.8672Mhz which is half that.
I'm guessing that the crystal on an NTSC one will be different.

alex3331, 2025-05-10:
Depending on the region (PAL/NTSC) Sharp computers may have two time bases:
17.7344 MHz this is the dot clock for 640x200 video (MZ-800 only), 8.8672 MHz is the clock for 320x200 video and 3.547 MHz is processor clock

or 14.31818 MHz, with 7.15909 MHz dot clock for 320x200 (MZ-700 and MZ-1500) and 3.5795 processor clock

MZ-700 and MZ-1500 use  the same LSI M60719 for timing, ; the crystal can be swapped and there is PAL/NTSC switch pin. 
MZ-800 has different LCI which includes video converter (RGBI output)

alex3331, 2025-05-10:
MZ-1500 Crystal 14.3181   CPU 3.57957 MHz
MZ-700  Crystal 17.73495  CPU 3.54685 MHz
MZ-800 Crystal  17.73945  CPU 3.54687 MHz
as measured

IanSB, 2025-05-10:
@alex3331 

Here are some profiles to test:
[Sharp_profiles.zip](https://github.com/user-attachments/files/20140181/Sharp_profiles.zip)

Unzip and put the three folders in:
\Profiles\6-12_BIT_RGB\Camputers_+_Oric_+_Sharp
on the Sd card then reboot the Pi zero after which they should be available in the select profile menu

These profiles are all set up for separate sync rather than composite sync so you will have to make a revised cable using those signals.
If you also connect the intensity bit, such a cable will then work on all three machines (The intensity bit will be ignored on the 700 & 1500)

New cable connection details are here
https://github.com/hoglet67/RGBtoHDMI/wiki/Cables#sharp-mz700--mz1500-3-bit-rgb-ttl--sharp-mz800-4-bit-rgb-ttl

There are separate profiles for the MZ700 and MZ1500 although they should work with with either but that is to allow separate calibrations to be stored if required.

The MZ800 profile is set up for RGBI mode at 640x200 with the others set up for 320x200.
The 50Hz/PAL profiles should work but the 60Hz/NTSC profiles may need some tweaking as they contain some guesswork
The 50Hz or 60Hz sub-profiles should be detected automatically.

At a minimum you will need to run the auto calibration.
A flashing screen indicates something is wrong with the profile timing.

Can you post the source summary page from your 800 PAL and 1500 NTSC machines.

alex3331, 2025-05-10:
Thanks
Need some time to rewire the cables, probably get some DIN PLUGS. Will post everything when ready

alex3331, 2025-05-11:
I unpacked the zip and have 3 folders - MZ-700, MZ-800 and MZ-1500. Should I put them into \Profiles\6-12_BIT_RGB\Camputers_+_Oric_+_Sharp  without transfering files to root?

IanSB, 2025-05-11:
@alex3331 

Yes put those three folders in the Camputers+Oric+Sharp sub-folder

alex3331, 2025-05-11:
First test - MZ-1500 with subprofile 60 Hz works perfectly. Will test two others later today and let you know. Thank you for your help
Let me know if you need additional data.
Alex

IanSB, 2025-05-11:
@alex3331 

>Picture is perfect, but there is green snow

Have you run the auto calibration? (long press of SW3 or menu option)

Can you post a screencap of the source summary page: 
Go to the info & help menu and select source summary, then Press SW2 & SW3 (up & down) together which will screencap the image.
Also screencap the screen without the menu by pressing SW2 (middle button)
The files will be in /captures on the sd card

alex3331, 2025-05-11:
Tested MZ-800. It turns out that MZ-700/50 Hz fits perfectly here . Both resolutions 640x200 and 320x200 look fine. I still need to check 16-color palette(intensity bit), but have no time today. On the other hand, profile MZ-800 does not provide a stable picture neither with 50 or 60 Hz on MZ-800 machine.
I had to swap wires because my DIN connector was mirrored; so now on MZ-700 there is red snow, not green. I will do screen caps tomorrow and send you. I did calibration, but I will repeat the whole thing tomorrow before sending screencaps to you.

IanSB, 2025-05-11:
@alex3331 
> Tested MZ-800. It turns out that MZ-700/50 Hz fits perfectly here . Both resolutions 640x200 and 320x200 look fine. I still need to check 16-color palette(intensity bit), but have no time today. On [...]

The info I found online suggested that the MZ-800 had positive going sync pulses and the MZ-700 had negative going sync pulses so I configured them that way but if the MZ-800 gives a stable picture with the MZ-700 profile and doesn't work with the MZ-800 profile (it would be shifted to the right and jittery) then it seems that info is incorrect and the MZ-800 uses the same sync polarity as the MZ-700.

Here is a revised MZ-800 profile with the sync polarity settings changed to be the same as the MZ-700 for you to test:

[New-MZ-800.zip](https://github.com/user-attachments/files/20151664/New-MZ-800.zip)

Replace the Sharp_MZ-800_(HVSync) folder on the SD card with this new one and try again.

You must also delete the "/Saved_Profiles" folder in the root of the SD card (if it exists) as that might contain old settings that would override this new profile.

Note the MZ-700 profile cannot work fully with the MZ-800 because it only samples and displays 320 pixels per line so 640 pixel mode will be missing every other pixel plus it cannot display the intensity bit. Only the MZ-800 profile will sample 640 pixel and display the intensity setting.

>so now on MZ-700 there is red snow, not green

A couple of suggestions:

1. Check that you have actually connected the correct wire (wire 7) from the 12 way connector to Pin 6 on the DIN
2. Check that you haven't swapped the wires to pins 1 and 6 on the DIN as that would connect composite video to the red input on the MZ-700

IanSB, 2025-05-12:
@alex3331 
>Do you need me to do anything else with MZ-800? 

Have you confirmed that the intensity bit is working for two different levels of each colour?

Did you run an auto calibration? That will optimise the sample timing and the saved result will be in /Saved_profiles on the SD card so if there is an updated file in there, please post it and I will use that as the default in the next release.
(This applies to all the other profiles in PAL or NTSC)

Are you able to test this in NTSC too?
(This uses a different sub-profile which might need some tweaking in the geometry menu to centre the image)

alex3331, 2025-05-12:
Intensity bit is working - I see all 16 colors in a test program. Let me run an auto calibration .  Here it is. Will test in NTSC later. 

[Sharp_MZ-800_50Hz.txt](https://github.com/user-attachments/files/20161824/Sharp_MZ-800_50Hz.txt)

alex3331, 2025-05-12:
On MZ-700. After multiple calibrations I see a good improvement; still some red snow, very minimal is observed when R channel is active. On a black or blue screen with only a a few white letters it is almost invisible. In one of the games the rightmost column is red and it has some blinking artifacts and more snow. I attach a few screenshots

[image]
[image]
[image]

IanSB, 2025-05-12:
@alex3331 

It looks like the red channel is noisy on your system for some reason. My MZ-700 doesn't seem to have any problem with red noise but I don't really have any software to run on it yet.

Try changing the multiplier from x8 to x16 in the sampling menu then re-running the calibration. This will make the calibration more accurate and it might be able to tune out the noise.

alex3331, 2025-05-12:
It was a faulty ground on video cable. No snow now; again thanks a lot for your help. You mentioned you have MZ-700 - I designed a backplane with RAM card, Flash ROM reader/writer and floppy controller. Can share if you are interested.

IanSB, 2025-05-13:
@alex3331 
>It was a faulty ground on video cable

Strange that it only affected the red channel but glad you sorted that out.

>I designed a backplane with RAM card, Flash ROM reader/writer and floppy controller. Can share if you are interested

I might be, do you have any more details or photos?

I think the PAL profiles are all sorted out now, so it is only the 60Hz ones to finalise.
You mentioned above that the MZ-700 and MZ-1500 worked at 60Hz but I would like to see a source summary screencap from one of those to confirm the timings are exact plus can you post a calibrated saved profile at 60Hz so I can make that the default.

The 60Hz MZ-800 hasn't been checked yet.

Also the pixel aspect settings in the profiles I posted for 60Hz are not correct:
for the 60Hz MZ-700 and MZ-1500 the Pixel Aspect in geometry should be changed to: 
```
H Pixel Aspect = 4
V Pixel Aspect = 5 
```
for the 60Hz MZ-800 the Pixel Aspect in geometry should be changed to: 
```
H Pixel Aspect = 2
V Pixel Aspect = 5
```
This will result in taller video the same as when viewed at 60Hz on a normal TV

alex3331, 2025-05-13:
per your suggestion tried to switch MZ-700 into 60 Hz mode. It can be switched to this subprofile, but no matter how I save it, when I press menu again - it shows 50 Hz subprofile. 4:5 ratio works and does not influence picture quality. Maybe I need more detailed instructions how to do this

IanSB, 2025-05-13:
@alex3331 
>It can be switched to this subprofile, but no matter how I save it, when I press menu again - it shows 50 Hz subprofile

That means the timing is wrong in the 60Hz profile so I need to see the source summary page when outputting 60Hz
To force the 60hz sub profile, set the "Auto Switch to "Off". Then you can manually switch between 50 and 60Hz sub-profiles

alex3331, 2025-05-13:
auto switch off - here is MZ-700 60 Hz. Picture has a lot of artifacts, the display window is down, last line is hardly visible. 
Here is the source

[image]

IanSB, 2025-05-13:
@alex3331 

It says frame rate: 50Hz so the computer is outputting 312 line 50Hz PAL timing not 60Hz NTSC timing.
I need to see the source summary when the computer is outputting 60Hz NTSC timing.
Note the Source summary page doesn't live update so the computer must already be outputting 60Hz before selecting the menu.