hoglet67/RGBtoHDMI #440: IBM 3270 PC/XT

koliha, 2025-07-17:
I've been trying to get RGBtoHDMI working with a couple of old IBM 3270 PC's that I have.  You can kind of see things when I select an MDA or Hercules profile, but no matter how much I play around with line/clock settings, I can't seem to get it going.  Using the latest stable release.  I'm at a loss and have been tinkering for a few days now.  Any help would be greatly appreciated!

This is what Hercules/MDA look like:
[image]

Here is a source summary w/defaults:
[image]

Here is a source summary after some tweaks:
[image]

If I create a custom profile and let it detect some settings and reboot, it's a little worse:
[image]

Here is what it looks like after using "Analyse Timing" when creating a custom profile:
[image]

And here is more information on the 3270 PC, including the display adapter.  I'm using the single card setup right now on both of the systems.  I believe the refresh rate/output was not standard CGA/MDA.  This system was originally paired with a special monitor from IBM, made by Zenith (I have a functional one of these too):  https://www.seasip.info/VintagePC/5271.html

I also have an oscilloscope and a logic analyzer that I can use if needed.

IanSB, 2025-07-18:
The screenshots show measured h sync timing as 66489ns and 71891ns which are way off the expected values which would be 53763ns for 18.6Khz and 41666ns for 24Khz and vertical timing is shown as 35Hz instead of the expected 50Hz or 63Hz.
Those timing values are measured directly from the sync pulses so there are several reasons why that might be the case:

1. The connecting cable might be faulty and not passing the sync signals to the converter (you could check that with a multimeter)
2. The sync signals are output on non standard pins (The RGB signals are on non standard pins)
3. The voltage levels of the sync signals are non standard (e.g. 1v rather than TTL)
4. The video card is faulty

Some sort of pulse is being received otherwise you would get no sync detected but it could be video signals as suggested above or just crosstalk. Looking at the signals with an oscilloscope is the best way to confirm that the sync signals are present on the right pins.

koliha, 2025-07-18:
I'm not sure what scale or timing you are looking for regarding the scope, so I've captured several images here.  I compared signals directly from the port on the computer vs through the pins on the cable and voltages/signals look the same.  Pics from both here.  If there is some specific timing that you need to see, please let me know and I can check it.

Vsync seems to have three voltage levels, 0.5, 1.8, and 3.5v.
Hsync seems to be producing 3.5v peaks.

Hsync signal, through cable, 100ms:
[image]

Hsync signal, direct to port, 100ms:
[image]

Vsync signal, through cable, 100ms:
[image]

Vsync signal, direct to port, 100ms:
[image]

hsync signal direct to port, 10ms:
[image]

vsync signal direct to port, 10ms:
[image]

both signals direct to port, 100ms:
[image]

both signals, stacked, 500us:
[image]

Thanks again for the assistance.

IanSB, 2025-07-18:
Neither of those look like sync signals. 
The green one looks like video so probably the main video or intensity signal
The yellow one looks like crosstalk so probably unconnected or maybe an input.

Looking back at your initial screencaps, there is a rolling solid horizontal white line so that might be the vsync signal connected to either intensity or video (pin 6 or 7) with the other one being part of the actual video as you can see some text.
Hsync may be on one of the other pins 1 - 5 so you need to probe all those other pins with the scope.

Hsync will be a pulse of about 4us repeating every 54us or 42us depending on mode
Vsync will be a pulse of about 40us - 150us repeating every 20ms or 15ms depending on mode

You might want to check that pin 1 is actually ground too by testing continuity between that pin and some known ground point.

koliha, 2025-07-18:
On the original screencaps you can read text on the screen.  

C>path c:\dps
??
??
Bad command or filename
C>prompt $p$g
C:\> ???
C:\>

[image]

Pin 1 was used for the ground connection of the oscilloscope leads and behaved like a ground input.  I have an IBM monitor here that I can hook it up to and see the output just fine as well (it just has a horizontal adjustment coil that is on it's last leg so I'd rather not use it for long periods is all).  Using known good + tested hardware.  These 3270 PC systems have been known (for 40+ years) to work with standard MDA monitors using no special adapters, so the idea that the pins are completely non-standard seems like a bit of a stretch.

I'll pull the scope back out and will test a little further, using something closer to the timing you have indicated above to see if I can more clearly illustrate things are functional.

IanSB, 2025-07-18:
Looking at these screencaps from the video and assuming that you are using the MDA profile in the latest beta:

[image]

[image]

The text is being shown at two different widths which implies two different pixel clock rates so one possibility is that the mono / colour switch input is being affected in some way such that it is continually switching between mono and colour mode which would have different pixel clock rates and that would presumably mess up the sync signals too.

This might be caused by the way the converter is terminating its inputs.

BTW Have the converter and CGA cable been confirmed as working on another system?

IanSB, 2025-07-18:
To test this theory you will either have to disconnect pin 4 for mono (preferably at the PC end so that crosstalk is eliminated) or ground pin 4 for colour.