hoglet67/RGBtoHDMI #446: Flashing display on PPC640

roddersuk, 2025-08-09:
I've managed to get an LCD display up and running using an RGBtoHDMI on my Amstrad PPC640.
I've tweaked the profile to get the display to fill the screen and iron out the glitches.
However, it still complains that genlock is inhibited and as soon as I close the RGBtoHDMI menu the display starts to flash.
Nothing I've tried manages to obtain genlock so any pointers would be great.
Here's my profile:

[Custom_Profile_0_.txt](https://github.com/user-attachments/files/21700922/Custom_Profile_0_.txt)
and my log

[Log.txt](https://github.com/user-attachments/files/21700944/Log.txt)

roddersuk, 2025-08-10:
I've made a bit of progress. 
I started again with the CGA (IBM_New) profile.
I sorted the genlock problem by setting the Resolution to 1024x768 and Refresh to 60Hz.
I tweaked the geometry to eliminate glitches and fill the screen but the display still flashes when I exit the menu.
Here's the latest profile:

[CGA_(IBM_New).txt](https://github.com/user-attachments/files/21704053/CGA_.IBM_New.txt)
and log

[Log.txt](https://github.com/user-attachments/files/21704113/Log.txt)

I'm guessing that this line indicates the problem but I don't know what to do about it:

`Timing exceeds window: H=63694, V=16687947, Lines=262, VSync=0`

roddersuk, 2025-08-31:
Well  I swapped out the 6845 but no joy, still the same behaviour.
I'm confused because all the measurements I take with the stock CGA profile seem to be right but the image is shifted across the screen and it flashes. Could the RGB signals be shifted from the sync? Its too much for my poor brain. I haven't recapped the board but the caps look fine and there are no electrolytics in this area anyway. It's looking more and more like the custom video chip being at fault but I'm not going to find one of those anytime soon.

IanSB, 2025-09-02:
Is the noise burst on the +5v supply of the 6845?
Can you post a screenshot of the hsync pulse and noise on your scope?
Do you get the same noise on the HSync out of the 6845 and the 9 way DIN?

What happens if you disconnect the reset input on the 6845? (i.e. bend out pin if socketed)

IanSB, 2025-09-04:
@roddersuk 

The source summary measured timings look correct with 63694ns per line and 262 lines which is CGA standard.
The only thing that looks wrong is the detected sync type which is showing as -H+V when the H syncs shown on the scope traces are postive going (+H).
What to the hsyncs look like on a longer timebase such as one or more frames? 

Various settings in the CPLD like sync polarity inversion are controlled by an internal register which is set serially by 3 signals so check that the following are connected and make sure the solder connections on those CPLD pins are good and not shorted to adjacent pins:

40 way header pin 28 to CPLD pin 6
40 way header pin 27 to CPLD pin 7
40 way header pin 38 to CPLD pin 44

You can see those connection details on the schematic:
https://raw.githack.com/hoglet67/RGBtoHDMI/master/Kicad_6Bit/v4/rgb-to-hdmi.pdf

Did you have to cut the 3 solder jumpers on the underside to initially program the CPLD?
If so did you remake them properly as they isolate the above connections.