hoglet67/RGBtoHDMI #453: Yamaha CX5MF (analog RGB) : no sync detected

localhero44, 2025-09-27:
Hi,

So I've bought the analog board and a cable with DIN 8 (for Sinclair QL) from cOpperdragon as I can't build it myself.
I've modified the Sinclair QL DIN 8 pins to match with what I think is needed for the Yamaha CX5M (based on this link : https://www.msx.org/wiki/RGB_(8-pin_DIN_45326)

Yamaha CX5M DIN pinout :

[image]

So I had these modifications in mind :

[image]

Here are 3 pins that I think I had to move :

[image]

First try with the Yamaha CX5MF : I see some horizontal lines but there is clearly a problem.
I get "no sync detected", which is not good obviously.

May I have something wrong with the pin numbers and then with my modification on the cable ?

Regards
David

IanSB, 2025-09-28:
>I get "no sync detected", which is not good obviously.

What profile are you using as a starting point for your custom profile? The above data states that sync is 0.3v p-p so if your base profile is using a higher sync voltage you need to adjust the sync level in the sampling menu to about 0.1v to get locked sync.

Is there any technical info on the timing of the yamaha video output? (resolution, pixel clock frequency etc)

localhero44, 2025-09-29:
Hi Ian,
Thank you for the answer.
I have tried so many things last Saturday, I am not sure to remember but I think I've started with the Sinclair QL profile. I can restart from a unmodified profile if needed.
After my initial post, I've tried the latest beta and got an almost stable image after a lot of trials and errors (still some pixels "moving").
But I've modified parameters I don't understand 🤣

I am not at home now, I could give you more details tonight.
I haven't found the pixel clock frequency in the service manual. The resolution is graphic mode 256 × 192 pixels, 16 colors. Standard resolution for all MSX computers.
The video display processor is TMS9918A.
Note the CX5MF (french version) is the only CX5M with a RGB output (through this DIN connector).

localhero44, 2025-09-29:
My "best" result so far (and image not stable anymore)

[image]

localhero44, 2025-10-01:
Oh, I thought the analog RGB would be enough as I had seen MSX in the list of computers. But again I am not an expert in video signals and I am completely lost.
I have to say that I have discovered the informations as you asked them.
I hadn't got such problems with digital RGB (for my Oric for example,  I got a perfect image at first try 😀).
And I didn't understand what was lumacode mentioned several times.

But I see a RCA cable with this digitizer, I have an analog RGB signal through a DIN 8. You speak about YUV output, but the CX5MF has analog RGB output.
Am I mixing things?

I had chosen RGB to program the CPLD.
Now I am not sure it was right.

I haven't found the exact technical manual to know the chip that provides this RGB analog output but I will open the computer to try to see 😉

localhero44, 2025-10-01:
I have taken some pictures.
The VDP chip is a TLS9929A

[image]

IanSB, 2025-10-03:
The TMS9929A has a YUV output so it must be getting converted to RGB by external circuitry

If you connect the analog RGBtoHDMI directly to the YUV pins on the chip you can try the TI99/4A profile

Connections would be as follows:

## TI99/4A
| RGBtoHDMI 6 way IDC | TMS99XX chip | Notes |
| --------- | ------------- | ----- |
| Pin 1  GND | Pin 12 | |
| Pin 2  SYNC | Pin 36 | Same as Y|
| Pin 3  U/BLUE | Pin 35 | |
| Pin 4  Y/GREEN | Pin 36 | |
| Pin 5  V/RED | Pin 38 | |
| Pin 6  +5V| Pin 33 | Powers RGBtoHDMI from the source, do not use a power supply |

Select the "U7 TI-99-4a 50Hz" YUV profile from the "Experimental" folder (not the lumacode one)

You will probably need to make small adjustments to the voltage levels in the sampling menu.